grblHAL/STM32F4xx #215: [Sienci SLB] Duty cycle offset on spindle PWM signal

johannsienci, 2025-01-28:
Hi there, Johann from Sienci Labs here, big fan of the project :)

I've found an issue with the spindle PWM signals using the SLB, where the duty cycle of the output signal is offset from the target by a fixed amount (added about ~40us in the example below, but varies from board to board), this amount applies to all target duty cycle and exists across all carrier frequencies. Laser PWM also has the same issue in my testing.

**1kHz Carrier - 10% target PWM -> 14% actual**
[image]

**1kHz Carrier - 50% target PWM -> 54% actual**
[image]

**1kHz Carrier - 90% target PWM -> 94% actual**
[image]

**2kHz Carrier - 10% target PWM  -> 18% actual**
[image]

**2kHz Carrier - 50% target PWM -> 58% actual**
[image]

**2kHz Carrier - 90% target PWM -> 98% actual**
[image]

terjeio, 2025-01-28:
It is a hardware issue, the optocouplers used has a very long turn off time. You may be able to reduce that somewhat by increasing the load in the emitter of the phototransistor.
When I designed my [Morpho board](https://github.com/terjeio/CNC_Breakout_Nucleo64) I selected a[ high frequency optocoupler](https://www.vishay.com/en/product/81587/) to mitigate this, I could not get precise timing with the more common ones - IIRC I tested with a 357 or similar type.

If you scope both the PWM output on the spindle header and the indicator LED you can see it.

johannsienci, 2025-01-28:
Thanks for the help @terjeio .

Just scoped it quickly and indeed the delay comes from the EL357N's switching time. We have workarounds for it and I'll let Chris know of this little quirk on the board.

Marking this closed.

[image]

closed 2025-01-28 by johannsienci