grblHAL/STM32F4xx #218: PWM timer configuration error

calabr, 2025-03-01:
Hi @terjeio,
  looks like it's wrong definition of WPM timer in pwm.c
[pwm_pin](https://github.com/grblHAL/STM32F4xx/blob/ed6b509659558189b1a3f459b8dca4b66ecf26bc/Src/pwm.c#L58-L61) structure - PB2 are not mapped to any timer in STM32F4 CPU

proposing also add instead of  [#if !IS_TIMER_CLAIMED(TIM2_BASE)](https://github.com/grblHAL/STM32F4xx/blob/ed6b509659558189b1a3f459b8dca4b66ecf26bc/Src/pwm.c#L49C1-L49C33))
```{   // !!! PB1 TIM1_CH3N
		    .port = GPIOB, .pin = 1, .timer = timer(1), .ccr = &timerCCR(1, 3), .ccmr = &timerCCMR(1, 2), .af = timerAF(2, 1),
		    .en = timerCCEN(3,N ), .pol = timerCCP(3,N ), .ois = timerCR2OIS(3,N ), .ocm = timerOCM(2, 3), .ocmc = timerOCM(2, 3)
	},
#if !IS_TIMER_CLAIMED(TIM2_BASE)
	{   // !!! PA0 TIM2_CH1
	    .port = GPIOA, .pin = 0, .timer = timer(2), .ccr = &timerCCR(2, 1), .ccmr = &timerCCMR(2, 1), .af = timerAF(2, 1),
	    .en = timerCCEN(1, ), .pol = timerCCP(1, ), .ois = timerCR2OIS(1, ), .ocm = timerOCM(1, 1), .ocmc = timerOCM(1, 1)
	},
	{   // !!! PA1 TIM2_CH2
	 	.port = GPIOA, .pin = 1, .timer = timer(2), .ccr = &timerCCR(2, 2), .ccmr = &timerCCMR(2, 1), .af = timerAF(2, 1),
	 	.en = timerCCEN(2, ), .pol = timerCCP(2, ), .ois = timerCR2OIS(2, ), .ocm = timerOCM(1, 2), .ocmc = timerOCM(1, 2)
	},
```
Could you clarify - is it mandatory to create #IF statement to surround PWM settings like this [#if !IS_TIMER_CLAIMED(TIM2_BASE)](https://github.com/grblHAL/STM32F4xx/blob/ed6b509659558189b1a3f459b8dca4b66ecf26bc/Src/pwm.c#L49C1-L49C33)) ?
 Why TIM2 and TIM3 are checked, but TIM1,TIM9,TIM11  are not ?
If the same port number can be mapped to several timers - is it safe to describe it for both in pwm_pin - like PA0 can be TIM2_CH1 or 
TIM5_CH1

terjeio, 2025-03-02:
> PB2 are not mapped to any timer in STM32F4 CPU

[image]

> Could you clarify - is it mandatory to create #IF statement to surround PWM settings

Not mandatory, IMO but it is better to get compile time errors instead of run time failures.

> Why TIM2 and TIM3 are checked, but TIM1,TIM9,TIM11 are not ?

Because not all are defined in [driver.h](https://github.com/grblHAL/STM32F4xx/blob/9b272398c8e3a748e1a746b313a3e28d26ea48d9/Inc/driver.h#L203-L257) (and some in a few map files) per now. And it takes time to configure and cross check every possible combination. So improvements is possible and can be made when the need arises.

calabr, 2025-03-02:
Which ST document are you using? 
Checked in [32F411](https://www.st.com/resource/en/datasheet/stm32f411ce.pdf)
[405/407](https://www.st.com/resource/en/datasheet/dm00037051.pdf)
[32F479](https://www.st.com/resource/en/datasheet/stm32f479ai.pdf)

terjeio, 2025-03-02:
446 - my main development platform. As you can see timer assignments is far from easy to implement in a perfectly flexible way...

calabr, 2025-03-02:
you right 446, are bit differ from other. Sorry to disturb

closed 2025-03-02 by calabr