grblHAL/STM32F4xx #220: X-Axis Drifting After Updating to grblHal Core 20250223

WhatsUGonnaDo, 2025-03-05:
Hi, 
I've run into a serious issue with X-position drift after updating my firmware from the stable grblHal Release 20231229 to the beta Release 20250223 The working coordinates are shifting unexpectedly, and by the end of a machining operation, the X position is completely off—even though no alarms are triggered.

What I’ve Checked & Tested:

 No loose couplings, slacks, or improperly tightened BK/BF blocks.
 No backlash compensation is enabled.
 Using Closed Loop CL57T V4.1 drivers from StepperOnline, which have been working fine so far.
 Adjusted parameters like pulse width and pulse delay—no real difference.
 Upgraded to the latest ioSender version 2.0.45, which resolved jog command issues.

Confirming It’s a Firmware Issue:

Downgraded back to the stable 20231229 → Machined successfully with no issues.

Upgraded again to 20250223 → Identical X-axis drift problem.

Downgraded back to 20231229 → Machined successfully once again 

Additional Notes:
The toolpath is a challenging one—a large adaptive operation with many small movements, running at 2800mm/min—but this was never an issue with the older stable release.
I've also sent a Contour OP along the same geometry but there’s no drift in a simple contour.  

I’ll be attaching the G-code file if anyone wants to emulate it

[GCode File & Settings ](https://we.tl/t-sjpFGNKXs4)

terjeio, 2025-03-05:
@Robin-SeaThrough Thanks for this, it gave me a clue for where to look. I have changed handling of the SPI clock when switching between the SD card and the W5500 ethernet chip - and introduced a bug in the process. I also changed the default SD card clock speed.

I have fixed the bug and changed the default speed back to the previous value and made it overridable, currently I have [overridden it for the SuperLongBoard](https://github.com/grblHAL/STM32F4xx/blob/939ff5615291b947a5496edff3ae204e71342a2f/boards/longboard32_map.h#L57-L58) to speeds that works for the selection of cards I have except one which I cannot get to work whatever speed combination I try. For some reason that card works with the iMXRT1062 driver but IIRC that uses a four lane SDIO interface...

So the issue is likely related to the STM32F4xxx driver, and possibly also the FlexiHAL hardware since both the SD card socket and the ethernet card are on a module.

WhatsUGonnaDo, 2025-03-05:
Sorry for the delay @terjeio , here is the settings.txt file and also the gcode for what's worth, just need to change the gcode extension back to .nc as you mentioned. 

I'm glad Robin was able to offer a more clear explanation of the events so you could get there quickly

Indeed I forgot to mention I was using the 20231229 version without the SD Card/Ethernet implementation prior and once I upgraded to 20250223 I did so with that variant as @Robin-SeaThrough mentioned. 

If there's anything else you may need, .F3D file from the Gcode etc please let me know. 

[settings.txt](https://github.com/user-attachments/files/19089541/settings.txt)

[1001.txt](https://github.com/user-attachments/files/19089557/1001.txt)

terjeio, 2025-03-05:
Can you try a firmware without ethernet/SD card enabled? Same issue? If not the fix committed may work.

Robin-SeaThrough, 2025-03-05:
@terjeio Thank you for the very quick response on this issue!

The changes you made, was that only in the STM32F4xx driver or the Core as well? I saw that you posted a new build of the Core earlier today :)

Would you expect the drifting issue to only affect the builds that have SD card and Ethernet enabled and/or the module plugged in? 

I still haven't seen the drifting in my machine, atleast nothing more than can be explained by general tolerances and less than ideal fixturing. I'll try to find time to run a test program tomorrow. 

It's a bit of a shame that we can't (atleast not fully/easily afaik?) use the Webbuilder for building firmware for the FlexiHAL, but I've that there's some work going in regards to that :)

I probably just have learn how to build with Platform.io 😅

terjeio, 2025-03-05:
> The changes you made, was that only in the STM32F4xx driver or the Core as well?

Only the driver. The core changes were for a different issue.

> Would you expect the drifting issue to only affect the builds that have SD card and Ethernet enabled and/or the module plugged in?

Possibly - the max step rate will be somewhat lower with ethernet/SD card enabled and in use simply because the MCU load increases. And slower SPI clock speeds increases the time waiting for SPI data transfers to complete - even as DMA is used. DMA reduces the MCU load so it is still beneficial.
This reminds me that I should check interrupts from the W5500 chip - I suspect it generates interrupts for every packed received even if it they are not for controller. So ideally the controller should be connected to a separate ethernet port?

I am running a test now with a Teensy 4.1 wired to my machine simulator which counts the step pulses. When finished I will see if steps are lost.

When the PR for adding flexiHAL is merged I'll add WebBuilder support - should not be long now.

terjeio, 2025-03-05:
Teensy 4.1 run completed successfully - step count ended up at 0, same as at the start. So the core is, IMO, ok.

WhatsUGonnaDo, 2025-03-05:
> Can you try a firmware without ethernet/SD card enabled? Same issue? If not the fix committed may work.

Yes of course, I’ve left the workshop but once I return tomorrow. 

Btw I haven’t noticed the issue on an Ethernet/SD card enabled firmware, the 20231229 version. 

Just to be clear I was using the “basic” 20231229 version.
When I downgraded from the newest back to 20231229 I did so to the variant that has the Ethernet/SD card enabled.

WhatsUGonnaDo, 2025-03-05:
I just went to check on a weird alarm I had half an hour ago in the sender, it was an error coming from the X axis driver. 7 blinks. 

“A red light flashing seven times on the StepperOnline CL47T V4.1 driver indicates a position tracking error. This means the difference between the motor's theoretical running angle and
its actual running angle exceeds one full rotation (default setting).”

I’m already downgraded back to 20231229. It was the first time it alarmed out and got an error from the driver, and was only on a jog movement. 

As I was telling @Robin-SeaThrougha few yours ago I even emulated the error once more with the newest build to be sure of what I was reporting and never got a drift with the old firmware. Otherwise wouldn’t have felt confident to report the error, but the alarm is on the axis that I’m seeing the drift.

WhatsUGonnaDo, 2025-03-08:
Hello Terje, sorry for the delay getting you an answer, unfortunately work got in the way. Today I set up to test the toolpath for a while, started with a test of the 20231229 with SD Card, everything alright, then tried the 20250223 without ethernet/SD card enabled and it drifted, tried it once again, drifted more, not once did I get an an alarm from the driver. After that, having the steppers hot, to rule out it being from heat creep or loss of torque, downgraded to the 20231229 without the SD Card and it didn't drift. I made sure not to interfere with feed overrides in any test, once again had issues everytime I upgraded, nothing else changed. 
In the meantime I spoke with Robin and asked him if he would mind running this toolpath on his PNC with 20250223 to see what he's findings were

Robin-SeaThrough, 2025-03-08:
As @WhatsUGonnaDo eluded to, I ran his Gcode program earlier today, with the FlexiHAL 1.0.1.3 firmware, without SD card or Network enabled, so that's with the 20250223 grblhal core.

And I got a pretty large deviation in the same way as @WhatsUGonnaDo, and in the same direction, X+. It did seem like I got a slightly bigger deviation, and I noted a slight deviation in Y as well. Got ~20mm in X and ~2mm in Y. That difference seems reasonable to me considering the Gcode. 

So yeah, the FlexiHAL 1.0.1.3 firmware should not be  used.

If I understand it correctly @terjeio, this error is due to the recent changes to the Core, with the SPI clock? And the newer builds should work with the older FlexiHAL drivers, or does it require different settings to override clock settings?

terjeio, 2025-03-10:
I see lost steps with the STM32F4xx driver when testing with the FlexiHAL map on a Nucleo-144 dev board, will try with a Nucleo-64 next with a different map. I have run several tests with iMXRT1062 (Teensy 4) and one with RP2350 (Pico 2) without detecting lost steps so I assume the core is ok - likely something in the driver is causing this.

> If I understand it correctly @terjeio, this error is due to the recent changes to the Core, with the SPI clock?

No, it is something else - I do not know what. The STM32F4xx test I ran was with SPI disabled. BTW the SPI clock change is in the driver, not in the core.

Robin-SeaThrough, 2025-03-10:
That sounds like progress to me, that you're able to replicate the lost steps with an STM32F4XX driver :)

A few of us have been discussing this over on the PrintNC Discord, I don't think I need to mention that you are more than welcome to join us Terje, but I do fully understand if you prefer to keep GRBLHAL discussions here on Github. 

If I had a spare FlexiHAL I'd send one to you for testing, but sadly I need the one I have :/

I don't want to speak for anyone else, but the the idea that it could somehow be related to the new Jerk or Acceleration Profiles setting have been brought up.
Jerk is not enabled in the FlexiHAL firmwares we have been using though.

@WhatsUGonnaDo did confirm that FlexiHAL 1.0.1.4 with Core 20250307 still has the drifting issue, and both him and me have tested 1.0.0.7 with Core 20240604 does not have the drifting issue at all.

terjeio, 2025-03-11:
I believe I've fixed it now... Sorry for causing problems.

Robin-SeaThrough, 2025-03-11:
> I believe I've fixed it now... Sorry for causing problems.

Thank you Terje! That sounds like something that cause a drift for sure, and it I believe I got a bigger deviation than @WhatsUGonnaDo and I have had a longer step delay. 
I do wonder if we actually need to have the step delay set the way we do, I did that based on previous experience with other systems and after reading the manual for the CL57T Driver that states that the Dir signal need to be atleast 2us ahead of the Pul signal. 
The PrintNC setup docs didnt mention $29 iirc, so I did start out with it set to 0, but I had problems during the first few startups where the 2 Y axis motors some times went in different directions during homing. Leading to an alarm and E-stop. 
I then added a Step delay of 8us and that solved the issue. Up until it actually happened once while I had the 1.0.1.3 firmware flashed. 

I think Ive read somewhere that GRBLHAL adds in a delay by default, but it was hard to find much info at all about $29.

WhatsUGonnaDo, 2025-03-11:
> I believe I've fixed it now... Sorry for causing problems.

Thank you Terje! Glad to know :)

terjeio, 2025-03-11:
> I think Ive read somewhere that GRBLHAL adds in a delay by default, but it was hard to find much info at all about $29.

Normally this is the case (likely due to the AMASS algorithm carried over from Grbl), and grblHAL drivers only adds the delay on direction changes that coincides with step pulses. The feed rate is low anyway when this happens so a few microseconds of delay does not matter.
What went wrong is that now and then interrupt processing elsewhere sometimes lasted so long that the step pulse interrupt for turning on the puls(es) were arriving too late. The change ensures that pulses are output anyway - even if somewhat later than they ideally should be. 

It would be nice if some of you would try/verify [Web Builder](https://svn.io-engineering.com:8443/?driver=STM32F4xx&board=Flexi-HAL) builds since I do not have a board to test with.

Robin-SeaThrough, 2025-03-17:
> Normally this is the case (likely due to the AMASS algorithm carried over from Grbl), and grblHAL drivers only adds the delay on direction changes that coincides with step pulses... 
> 

Thank you for the explanation Terje! Much appreciated! 
If I understand it correctly, its not a bad idea to use a bit of Pulse Delay to make sure there's always a sufficient delay between Dir and Step, but it probably doesnt need to be much larger than the minimum specced by the driver? 
It seems that things would be more sensitive when running short Step pulses/higher step frequencies? 

I see now that none of us reported back that the latest updates to the driver solved the drifting we were seeing, available in the FlexiHAL 1.0.1.6 firmware from Expatria, thank you very much for finding and fixing this issue!

I've had to focus on making parts, but I plan to try the Webbuilder and test it on the machine, possibly this week.