grblHAL/STM32F4xx #223: Fysetc Cheetah v2

DavMem, 2025-03-29:
I have this board: https://wiki.fysetc.com/Cheetah_Board_V20/#43-pin-definition
and am trying to get it to work with grblHAL.
I can compile and flas to the board through dfu-util or stm32cubeprogrammer. 
I have tried flashing the firmware directly or flashing the cheetah bootloader and letting the board install it.
Whenever the board is not in dfu mode however i am unable to communicate with it.
Do i need to set something specific? Does this board not work with grblHAL?
My platformio ini for it lloks like this:

[env:cheetah]
board = genericSTM32F401RC
board_build.ldscript = STM32F401RC_FLASH.ld
build_flags = ${common.build_flags}
lib_deps = ${common.lib_deps}
lib_extra_dirs = ${common.lib_extra_dirs}

in my_maschine i have just mapped the stepper pins and endstops.

terjeio, 2025-03-29:
You have modified the generic map?
If you want to communicate via USB you have to add:
 -D USB_SERIAL_CDC=1
And you may have to specify the crystal frequency, e.g for 8MHz (default is 25MHz):
 -D HSE_VALUE=8000000

DavMem, 2025-03-31:
Thanks, I had set usb serial in machine but didn't set the 8mhz. It works now

closed 2025-03-31 by DavMem