grblHAL/STM32F4xx #227: Serial Communication Issue in STM32F4XX Project

the-flying-fish-668, 2025-04-27:
I downloaded a clean ‘STM32F4XX’ project from GitHub and opened it using PlatformIO. I created a new environment laser_tmc2209(STM32F4xx) with the following configuration:

[env:laser_tmc2209]
##board = genericSTM32F407ZGT6
board = black_f407zg
board_build.ldscript = STM32F407ZGT6_Laser_FLASH.ld
build_flags = ${common.build_flags}
  -D WEB_BUILD
##-D BOARD_BTT_SKR_20
  -D BOARD_JIAN_MEI_LASER
  -D USE_HAL_DRIVER
  -D STM32F407xx
  -D HSE_VALUE=12000000
  -D USB_SERIAL_CDC=0
  -D PROBE_ENABLE=1
  -D TRINAMIC_ENABLE=2209
  -D TRINAMIC_UART_ENABLE=2
  -D BLUETOOTH_ENABLE=0
  -D NETWORKING_ENABLE=0
  -D ESTOP_ENABLE=1
  -D SPINDLE0_ENABLE=11
  -D N_SPINDLE=1
  -D DEBUGOUT=0
  -D DEBUG=1
  -D USE_FULL_ASSERT  # Enable HAL library assertions

lib_deps = ${common.lib_deps}
  eeprom

lib_extra_dirs = ${common.lib_extra_dirs}

upload_protocol = jlink
(This configuration is modified from [env:btt_skr_2_tmc2209], and the board header files are also based on it.)

The project compiles successfully in PlatformIO, which is great. However, after flashing it to my development board, I found that the serial communication is not working properly.

I made the following changes to the serial port configuration:

//#define SERIAL_PORT   1  // GPIOA: TX = 9, RX = 10   // Original serial port 1 definition (commented out)
#define SERIAL_PORT   6  // GPIOC: TX = 6, RX = 7     // Using serial port 6 on my board
I disabled USB_SERIAL_CDC since I’m using UART6, but the serial communication doesn’t work. The laserGRBL software log shows that G-code is being sent, but there’s no response from the serial port.

Interestingly, I wrote a separate demo in PlatformIO to test UART6, and it worked fine. I’m not sure what’s causing this issue in the main project.

I understand this might be a simple issue, but I’m still a beginner. I’ve checked everything I could think of:

HSE clock configuration
USART6 serial port settings
Main communication port definitions in the GRBL library
Properly disabled USB_CDC
Despite these checks, the problem persists. It might not even be a serial port issue itself—perhaps something else in the project configuration is interfering.

Could you please provide some suggestions? Thank you very much~

terjeio, 2025-04-27:
Just retested and confirmed working with a F446 Nucleo-144 board.

Do you have any conflicting use of the pins?
Or perhaps the board is not booting at all? Can you set breakpoints for debugging?

the-flying-fish-668, 2025-04-28:
Wow, I'm really happy to receive your reply. Thank you!

I was previously using Keil5 and wasn't very familiar with PlatformIO. Your suggestion reminded me that I could use breakpoint debugging in PlatformIO as well. During testing, I confirmed the program wasn't running properly, but the good news is that the breakpoint debugging actually helped identify the issue - there was an address access violation. I suspect this was caused by a mismatch between the memory allocation in the linker script and my development board. After consulting DeepSeek, I modified the original memory configuration.

[image]

After testing, I can now confirm that USART6 is successfully outputting debug information (though since I haven't modified any functional pin mappings yet, the debug messages show errors - but the current issue is resolved),  I don't quite understand why this is happening, I still need to delve deeper into it

By the way, could you advise which file I should modify for pin mapping? Can I make the changes directly in the header files under the boards directory?

terjeio, 2025-04-28:
If for a private custom board then just add _my_machine_map.h_ to the boards folder and [enable that](https://github.com/grblHAL/STM32F4xx/blob/00d9b13681660fa19c2871f8959e523eb2fea3fa/Inc/my_machine.h#L47).

the-flying-fish-668, 2025-04-28:
“OK, got it. Thank you for the reply!   I’ll close this issue now.”     ~❀~❀~❀~

closed 2025-04-28 by the-flying-fish-668