grblHAL/STM32F4xx #229: Fails to compile

6XoCtujg2C0gne, 2025-05-13:
Processing btt_skr_pro_1_1 (board: genericSTM32F407VGT6; platform: ststm32; framework: stm32cube)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F407VGT6.html
PLATFORM: ST STM32 (17.6.0) > STM32F407VG (128k RAM. 1024k Flash)
HARDWARE: STM32F407VGT6 168MHz, 128KB RAM, 1MB Flash
DEBUG: Current (stlink) External (jlink, stlink)
PACKAGES:
 - framework-stm32cubef4 @ 1.26.2
 - tool-ldscripts-ststm32 @ 0.2.0
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 97 compatible libraries
Scanning dependencies...
Dependency Graph
|-- boards    
|-- bluetooth 
|-- grbl      
|-- keypad    
|-- laser     
|-- motors    
|-- trinamic  
|-- odometer  
|-- fans      
|-- spindle   
|-- embroidery
|-- App @ 0.0.0+20250204161749
|-- Target @ 0.0.0+20250204161749
|-- Core @ 0.0.0+20250204161749
|-- Class @ 0.0.0+20250204161749
|-- App @ 0.0.0+20250204161749
|-- Target @ 0.0.0+20250204161749
|-- FatFs
|-- sdcard
|-- eeprom
Building in release mode
Compiling .pio\build\btt_skr_pro_1_1\lib45d\boards\Devtronic_CNC_Controller.o
Compiling .pio\build\btt_skr_pro_1_1\lib45d\boards\btt_skr_2.0.o
Compiling .pio\build\btt_skr_pro_1_1\lib45d\boards\flexi_hal.o
Compiling .pio\build\btt_skr_pro_1_1\lib45d\boards\longboard32.o
Compiling .pio\build\btt_skr_pro_1_1\lib45d\boards\st_morpo_dac.o
Compiling .pio\build\btt_skr_pro_1_1\lib9e9\bluetooth\hc_05.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\alarms.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\canbus.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\coolant_control.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\crossbar.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\errors.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\gcode.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\grbllib.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\ioports.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\kinematics\corexy.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\kinematics\delta.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\kinematics\maslow.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\kinematics\polar.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\kinematics\wall_plotter.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\machine_limits.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\messages.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\modbus.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\modbus_rtu.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\motion_control.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\ngc_expr.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\ngc_flowctrl.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\ngc_params.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\nuts_bolts.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\nvs_buffer.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\override.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\planner.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\protocol.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\report.o
Compiling .pio\build\btt_skr_pro_1_1\libdf2\grbl\settings.o
grbl\settings.c:90:33: error: 'DEFAULT_TOOLCHANGE_AT_G30' undeclared here (not in a function); did you mean 'DEFAULT_TOOLCHANGE_MODE'?
     .flags.tool_change_at_g30 = DEFAULT_TOOLCHANGE_AT_G30,
                                 ^~~~~~~~~~~~~~~~~~~~~~~~~
                                 DEFAULT_TOOLCHANGE_MODE
grbl\settings.c:91:39: error: 'DEFAULT_TOOLCHANGE_FAST_PROBE_PULLOFF' undeclared here (not in a function); did you mean 'DEFAULT_TOOLCHANGE_FEED_RATE'?
     .flags.tool_change_fast_pulloff = DEFAULT_TOOLCHANGE_FAST_PROBE_PULLOFF,
                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                       DEFAULT_TOOLCHANGE_FEED_RATE
grbl\settings.c:93:36: error: 'DEFAULT_KEEP_OFFSETS_ON_RESET' undeclared here (not in a function); did you mean 'DEFAULT_REPORT_INCHES'?
     .flags.keep_offsets_on_reset = DEFAULT_KEEP_OFFSETS_ON_RESET,
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                    DEFAULT_REPORT_INCHES
grbl\settings.c:97:27: error: 'DEFAULT_SOFT_LIMIT_PROBE_CYCLES' undeclared here (not in a function); did you mean 'DEFAULT_SOFT_LIMIT_ENABLE'?
     .probe.soft_limited = DEFAULT_SOFT_LIMIT_PROBE_CYCLES,
                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                           DEFAULT_SOFT_LIMIT_ENABLE
*** [.pio\build\btt_skr_pro_1_1\libdf2\grbl\settings.o] Error 1
==================================================================================== [FAILED] Took 24.11 seconds ====================================================================================

Environment      Status    Duration
---------------  --------  ------------
btt_skr_pro_1_1  FAILED    00:00:24.108
=============================================================================== 1 failed, 0 succeeded in 00:00:24.108 ===============================================================================

Previous version compile fine.

terjeio, 2025-05-13:
[grbl/config.h](https://github.com/search?q=repo%3AgrblHAL%2Fcore+DEFAULT_TOOLCHANGE_AT_G30&type=code) is out of sync or corrupted?

6XoCtujg2C0gne, 2025-05-13:
Yes it indeed was. Thanks!

Is there a list of error codes? I'm getting #79 after successfully connecting.

Andy

terjeio, 2025-05-13:
> Is there a list of error codes?

`$EE` (**E**numerate **E**rror codes)

closed 2025-05-13 by 6XoCtujg2C0gne