grblHAL/STM32F4xx #230: SKR PRO V1.2 Compiles and connects

6XoCtujg2C0gne, 2025-05-14:
I've compiled and connected but continue to get a ERROR: 79 message.

I am using this BTT SKR PRO V1.2 as a laser controller with single X axis and dual Y axis.

[env:btt_skr_pro_1_1]
# Untested and might not boot.  Please report issues at:
# https://github.com/grblHAL/STM32F4xx/issues
board = genericSTM32F407VGT6
board_build.ldscript = STM32F407VGTX_BL32K_FLASH.ld
build_flags = ${common.build_flags}
  # See Inc/my_machine.h for options
  -D BOARD_BTT_SKR_PRO_1_1
  -D TRINAMIC_ENABLE=2209
  -D ESTOP_ENABLE=1
  # 8MHz crystal
  -D HSE_VALUE=8000000
  
  -D HAS_BOOTLOADER=1
  -D USB_SERIAL_CDC=1
  -D TRINAMIC_UART_ENABLE=2
  -D SDCARD_ENABLE=1
  -D LASER_COOLANT_ENABLE=1
  -D LASER_OVD_ENABLE=1
  -D FANS_ENABLE=1

==========================================

CNCjs 1.10.5 [Grbl]
Connected to COM4 with a baud rate of 115200
GrblHAL 1.1f ['$' or '$HELP' for help]
client> $$
$0=5.0 (Step pulse time, microseconds)
$1=25 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=1 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$9=1
$10=0 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$14=0
$15=0
$16=0
$17=0
$18=0
$19=0
$20=0 (Soft limits enable, boolean)
$21=1 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=3 (Homing direction invert, mask)
$24=25.0 (Homing locate feed rate, mm/min)
$25=500.0 (Homing search seek rate, mm/min)
$26=250 (Homing switch debounce delay, milliseconds)
$27=4.000 (Homing switch pull-off distance, millimeters)
$28=0.100
$29=0.0
$30=255.000 (Maximum spindle speed, RPM)
$31=1.000 (Minimum spindle speed, RPM)
$32=1 (Laser-mode enable, boolean)
$33=5000.0
$34=0.0
$35=0.0
$36=100.0
$37=0
$39=1
$40=0
$43=1
$44=4
$45=3
$46=0
$62=0
$63=3
$64=0
$65=0
$100=100.00000 (X-axis travel resolution, step/mm)
$101=100.00000 (Y-axis travel resolution, step/mm)
$102=100.00000 (Z-axis travel resolution, step/mm)
$110=1000.000 (X-axis maximum rate, mm/min)
$111=1000.000 (Y-axis maximum rate, mm/min)
$112=500.000 (Z-axis maximum rate, mm/min)
$120=20.000 (X-axis acceleration, mm/sec^2)
$121=20.000 (Y-axis acceleration, mm/sec^2)
$122=10.000 (Z-axis acceleration, mm/sec^2)
$130=666.000 (X-axis maximum travel, millimeters)
$131=714.000 (Y-axis maximum travel, millimeters)
$132=200.000 (Z-axis maximum travel, millimeters)
$140=500
$141=500
$142=500
$150=16
$151=16
$152=16
$200=22.0
$201=22.0
$202=22.0
$210=50
$211=50
$212=50
$220=22.0
$221=22.0
$222=22.0
$338=7
$339=0
$341=0
$342=30.0
$343=25.0
$344=200.0
$345=200.0
$346=1
$370=0
$378=0.0
$379=0.0
$384=0
$386=0.0
$391=0.0
$394=0.0
$398=100
$480=0.0
$481=0
$483=0
$484=1
$486=0
$539=0.0
$650=0
$673=0.0
$676=3
ok
client> $X
error:79
client>
client> $X
error:79
client> $H
error:79
>
====================

Any ideas where to start?

terjeio, 2025-05-14:
> Any ideas where to start?

[Here](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings)?

6XoCtujg2C0gne, 2025-05-14:
Awesome, thanks!

closed 2025-05-14 by 6XoCtujg2C0gne