grblHAL/STM32F4xx #231: GrblHal on SKR PRO V1.2 motor movement

6XoCtujg2C0gne, 2025-05-16:
I seem to be having an issue with the motors only moving a 'bump' instead of a full rotation despite what is commanded.

My question/problem: Why are my motor's moving just a bump when commanded to move, say 5mm?

Board SKR PRO v1.2 with TMC2209 drivers
Configuration: Single X axis and dual Y axis. No Z axis
Configured as a laser.

Successfully compiled and uploaded using:
[env:btt_skr_pro_1_1]
board = genericSTM32F407VGT6
board_build.ldscript = STM32F407VGTX_BL32K_FLASH.ld
build_flags = ${common.build_flags}
  -D BOARD_BTT_SKR_PRO_1_1
  -D TRINAMIC_ENABLE=2209
  -D ESTOP_ENABLE=1
  -D HSE_VALUE=8000000
  -D Y_GANGED=1
  -D Y_AUTO_SQUARE=1
  -D HAS_BOOTLOADER=1
  -D USB_SERIAL_CDC=1
  -D SDCARD_ENABLE=1
  -D LASER_COOLANT_ENABLE=1
  -D LASER_OVD_ENABLE=1
  -D FANS_ENABLE=1

> M122
> [TRINAMIC]
>                       X       Y      Y2
> Driver          TMC2209 TMC2209 TMC2209
> Set current         500     500     500
> RMS current         490     490     490
> Peak current        692     692     692
> Run current       15/31   15/31   15/31
> Hold current       7/31    7/31    7/31
> CS actual          7/31    7/31    7/31
> PWM scale             9       9       9
> vsense          1=0.180 1=0.180 1=0.180
> stealthChop       false   false   false
> msteps               16      16      16
> tstep           1048575 1048575 1048575
> pwm
> threshold             0       0       0
> [mm/s]                0       0       0
> OT prewarn        false   false   false
> OT prewarn has
> been triggered    false   false   false
> off time              3       3       3
> blank time            1       1       1
> hysteresis
> -end                 -1      -1      -1
> -start                1       1       1
> Stallguard thrs       0       0       0
> DRIVER STATUS:
> stallguard     
> sg_result           432     238     464
> fsactive                               
> stst                  *       *       *
> olb                   *                
> ola                   *                
> s2gb                                   
> s2ga                                   
> otpw                                   
> ot                                     
> STATUS REGISTERS:
>  X = 0x80:07:00:C0
>  Y = 0x80:07:00:00
>  Y2 = 0x80:07:00:00
> ok
> 
> 

 $$
$0 = 10.0    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 1    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$8 = 0   
$9 = 1   
$10 = 7    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$14 = 70   
$15 = 0   
$16 = 0   
$17 = 0   
$18 = 0   
$19 = 0   
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 25.0    (Homing locate feed rate, mm/min)
$25 = 500.0    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 4.000    (Homing switch pull-off distance, millimeters)
$28 = 0.000   
$29 = 0.0   
$30 = 255.000    (Maximum spindle speed, RPM)
$31 = 1.000    (Minimum spindle speed, RPM)
$32 = 1    (Laser-mode enable, boolean)
$33 = 5000.0   
$34 = 0.0   
$35 = 0.0   
$36 = 100.0   
$37 = 0   
$39 = 1   
$40 = 0   
$43 = 1   
$44 = 0   
$45 = 1   
$46 = 0   
$62 = 0   
$63 = 0   
$64 = 0   
$65 = 0   
$100 = 100.00000    (X-axis travel resolution, step/mm)
$101 = 100.00000    (Y-axis travel resolution, step/mm)
$102 = 0.00000    (Z-axis travel resolution, step/mm)
$110 = 1000.000    (X-axis maximum rate, mm/min)
$111 = 1000.000    (Y-axis maximum rate, mm/min)
$112 = 0.000    (Z-axis maximum rate, mm/min)
$120 = 20.000    (X-axis acceleration, mm/sec^2)
$121 = 20.000    (Y-axis acceleration, mm/sec^2)
$122 = 0.000    (Z-axis acceleration, mm/sec^2)
$130 = 666.000    (X-axis maximum travel, millimeters)
$131 = 714.000    (Y-axis maximum travel, millimeters)
$132 = 0.000    (Z-axis maximum travel, millimeters)
$140 = 500   
$141 = 500   
$142 = 500   
$150 = 16   
$151 = 16   
$152 = 16   
$170 = 0.000   
$171 = 0.000   
$172 = 0.000   
$200 = 22.0   
$201 = 22.0   
$202 = 22.0   
$210 = 50   
$211 = 50   
$212 = 50   
$220 = 22.0   
$221 = 22.0   
$222 = 22.0   
$338 = 3   
$339 = 0   
$341 = 0   
$342 = 30.0   
$343 = 25.0   
$344 = 200.0   
$345 = 200.0   
$346 = 1   
$347 = 5.0   
$348 = 2.500   
$349 = 25.000   
$370 = 0   
$378 = 0.0   
$379 = 0.0   
$384 = 0   
$386 = 0.0   
$391 = 0.0   
$394 = 0.0   
$398 = 100   
$480 = 0.0   
$481 = 0   
$483 = 0   
$484 = 1   
$486 = 0   
$539 = 0.0   
$650 = 0   
$673 = 0.0   
$676 = 3   
ok

dresco, 2025-05-16:
Possibly your stepper enable inversion setting? I believe the default is inverted for each axis (so `$4=7` for a 3 axis build)..

```
$4: Invert stepper enable output(s) as axismask
Inverts the stepper driver enable signals. Most drivers uses active low enable requiring inversion.
```

6XoCtujg2C0gne, 2025-05-16:
Absolutely spot-on! Works perfectly. I appreciate your help!

--Andy

closed 2025-06-02 by terjeio