grblHAL/STM32F4xx #240: SLB_EXT/SLB Compilation issue with PIO (lwip\App\lwip.h:29:10: fatal error: lwip/opt.h: No such file or directory)

petervanderwalt, 2025-07-10:
Hi @terjeio 

I am trying to use PIO/VisualStudio to compile STM32F4xx for a Sienci Longboard/Longboard-EXT

I am using your platform.io from https://github.com/grblHAL/STM32F4xx/issues/193#issuecomment-2412859197 

However, I am getting an error regarding missing LWIP files etc 

Have made sure to pull recursively. 

I am willing to bet its just me doing something stupid again (: but I am hoping you can point me at the obvious I am missing. 

PS: I want to do this to test your pull request https://github.com/Sienci-Labs/grblhal-rgb-plugin/pull/3 (: 

```
PS C:\Users\User\Documents\GITHUB\SIENCI LABS\STM32F4xx> pio run
Warning! Ignore unknown configuration option `grblhal_driver_version` in section [env]
Processing f412vg_SLB (board: genericSTM32F412VG; platform: ststm32; framework: stm32cube)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F412VG.html
PLATFORM: ST STM32 (19.2.0) > STM32F412VG (256k RAM. 1024k Flash)
HARDWARE: STM32F412VGT6 100MHz, 256KB RAM, 1MB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-stm32cubef4 @ 1.26.2
 - tool-ldscripts-ststm32 @ 0.2.0
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 92 compatible libraries
Scanning dependencies...
Dependency Graph
|-- boards
|-- bluetooth
|-- grbl
|-- keypad
|-- laser
|-- motors
|-- odometer
|-- trinamic
|-- lwip
|-- plugins
|-- networking
|-- FatFs
|-- sdcard
|-- spindle
|-- eeprom
|-- webui
|-- Core
|-- Class
|-- App
|-- Target
Building in release mode
Compiling .pio\build\f412vg_SLB\lib3ca\lwip\App\lwip.o
Compiling .pio\build\f412vg_SLB\lib3ca\lwip\dp83848\Target\ethernetif.o
In file included from lwip\App\lwip.c:22:0:
lwip\App\lwip.h:29:10: fatal error: lwip/opt.h: No such file or directory
 #include "lwip/opt.h"
          ^~~~~~~~~~~~
compilation terminated.
lwip\dp83848\Target\ethernetif.c:23:10: fatal error: lwip/opt.h: No such file or directory
 #include "lwip/opt.h"
          ^~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\f412vg_SLB\lib8b7\networking\ftpd.o
*** [.pio\build\f412vg_SLB\lib3ca\lwip\App\lwip.o] Error 1
*** [.pio\build\f412vg_SLB\lib3ca\lwip\dp83848\Target\ethernetif.o] Error 1
networking\ftpd.c:53:10: fatal error: lwip/debug.h: No such file or directory
 #include "lwip/debug.h"
          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\f412vg_SLB\lib8b7\networking\ftpd.o] Error 1
========================================================================================== [FAILED] Took 5.58 seconds ==========================================================================================

Processing f412vg_SLB_EXT (board: genericSTM32F412VG; platform: ststm32; framework: stm32cube)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F412VG.html
PLATFORM: ST STM32 (19.2.0) > STM32F412VG (256k RAM. 1024k Flash)
HARDWARE: STM32F412VGT6 100MHz, 256KB RAM, 1MB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-stm32cubef4 @ 1.26.2
 - tool-ldscripts-ststm32 @ 0.2.0
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 92 compatible libraries
Scanning dependencies...
Dependency Graph
|-- boards
|-- bluetooth
|-- grbl
|-- keypad
|-- laser
|-- motors
|-- odometer
|-- trinamic
|-- lwip
|-- plugins
|-- networking
|-- FatFs
|-- sdcard
|-- spindle
|-- eeprom
|-- webui
|-- Core
|-- Class
|-- App
|-- Target
Building in release mode
Compiling .pio\build\f412vg_SLB_EXT\lib3ca\lwip\App\lwip.o
Compiling .pio\build\f412vg_SLB_EXT\lib3ca\lwip\dp83848\Target\ethernetif.o
In file included from lwip\App\lwip.c:22:0:
lwip\App\lwip.h:29:10: fatal error: lwip/opt.h: No such file or directory
 #include "lwip/opt.h"
          ^~~~~~~~~~~~
compilation terminated.
lwip\dp83848\Target\ethernetif.c:23:10: fatal error: lwip/opt.h: No such file or directory
 #include "lwip/opt.h"
          ^~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\f412vg_SLB_EXT\lib8b7\networking\ftpd.o
*** [.pio\build\f412vg_SLB_EXT\lib3ca\lwip\App\lwip.o] Error 1
*** [.pio\build\f412vg_SLB_EXT\lib3ca\lwip\dp83848\Target\ethernetif.o] Error 1
networking\ftpd.c:53:10: fatal error: lwip/debug.h: No such file or directory
 #include "lwip/debug.h"
          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\f412vg_SLB_EXT\lib8b7\networking\ftpd.o] Error 1
========================================================================================== [FAILED] Took 5.62 seconds ==========================================================================================

Environment     Status    Duration
--------------  --------  ------------
f412vg_SLB      FAILED    00:00:05.581
f412vg_SLB_EXT  FAILED    00:00:05.622
===================================================================================== 2 failed, 0 succeeded in 00:00:11.203 =====================================================================================
PS C:\Users\User\Documents\GITHUB\SIENCI LABS\STM32F4xx>
```

terjeio, 2025-07-10:
The lwip code has been moved/changed to support native ethernet - create a Web Builder build with the [link provided here](https://github.com/grblHAL/core/wiki/Compiling-grblHAL#platformio) to get the .ini file used and either use that or update yours. FYI the Web Builder uses the code from the PR so you may use the generated binary...

petervanderwalt, 2025-07-10:
Thanks @terjeio - should we still merge it though? 
And thank you for the updated info as well, knew I was being daft, but not reading the docs is exceedingly so (: - my apologies! 

Have a great day!

petervanderwalt, 2025-07-11:
Confirmed, LEDs works great.   Will ask the team to merge the PR too, thank you!

closed 2025-07-11 by petervanderwalt