grblHAL/STM32F4xx #252: Looking to make a driver from already available driver list

elekgal-bit, 2025-08-30:
I replaced pinout from skr_pro_v1_1_map.h  with pinout for FLY Mellow8 . Added pin definitions M6 , M7 under M5 (skr pro 1.1 is 6 axis ). 
    Didn't tryed to compile . 
   Q: Will this work ?  Or , i have to make more changes ?
  Compiled grbl for Mega boards , but never for STM32

terjeio, 2025-08-31:
It will work if the board uses the same F4 variant and crystal frequency. If not you will have to change the build configuration. You may also have to add a new clock tree setup if the variant is not currently supported. Do you have a link to the board?

elekgal-bit, 2025-08-31:
https://github.com/Mellow-3D/Fly-Super8 (I have this)

terjeio, 2025-08-31:
Ok, same procesor variant as the SKR, if the crystal is 8 MHz it should work. You will flash via the SD card (bootloader)?

elekgal-bit, 2025-08-31:
I know of 3 variants  :
Super8 : 32-bit ARM Cortex-M4 series 168 MHz, STM32F407ZGT6 chip , 1Mb flash

Super8 PRO : 32-bit ARM Cortex-M7 Kernel 550MHz, STM32H723ZGT6 Chip. 
https://mellow-3d.github.io/fly_super8_pro_h723_general.html

 I saw  with STM32H743ZIT6 cip to.

elekgal-bit, 2025-08-31:
https://mellow.klipper.cn/en/docs/ProductDoc/MainBoard/fly-super/fly-super8/flash/

terjeio, 2025-08-31:
Well, then you should just try with compiling a SKR bootloader version with the new map. If the crystal and the bootloaders are the same it could be that it just works.

elekgal-bit, 2025-09-01:
Suspended project for now . Just received the BTT Octopus Pro 1.1 with : STM32H723ZET chip , 512k flash , 550MHz .
   Generated firmware.bin with https://svn.io-engineering.com:8443/     

[image]

Inserted jumper on board : 

[image]

conected to pc via USB C

Followed this tutorial https://www.youtube.com/watch?v=VlCYI2U-qyM , from minute 3:20 

Disconnect from soft . Disconnect USB C . Remove jumper from BOOT pin. Reconnect to PC .
I'm using it with grblGRU . Works ! Have to remove alarms and properly config setups .

[image]

 Cheers !