grblHAL/STM32F4xx #255: enabling Tapping feature in 3 axis milling machine.

Bharathk-Gowda, 2025-09-05:
dear team,

i have encoder for my spindle 2000 puls per rotation, i need to get tapping feature in milling machine using GRBL firmware, kindly help me for enabling this.

Thanks.

terjeio, 2025-09-06:
Have you started adding the feature?
I do not have the hardware required to do it so I am currently not able to add it - sincd, IMO, it will require extensive testing.

Bharathk-Gowda, 2025-09-10:
no, I don't know any thing about embedded and programming, I am programmer for CNC machines, I am planning to build DIY CNC  3 axis milling machine, so I asked the tapping feature, kindly help me for enabling tapping feature, **give me a road map to enable tapping feature in GRBL, so that I can assign this task to some embedded freelancers**.

terjeio, 2025-09-21:
My roadmap is some time in the future - I have to get a spindle that I can use for tapping first, and then free time for coding and testing. And lots of taps for testing? I am sure I am going to break a bunch?
If you can get some freelancers to do it then that would be great.